first95/FRC2020 #11: Modify drivebase for sucker chassis

lindsayvallen, 2019-10-02:
Should be done in feature branch. Will need to wait until design is sufficiently far along to know what the software spec should be.

jwalthour, 2020-02-01:
I committed a draft of this functionality in f1581e66801b1903d826fa23084bc2aef4e40b6d.  I put it into the `DriveBase` subsystem for now; we should move it into its own subsystem if that ever becomes easier.

jwalthour, 2020-03-04:
Sucker chassis outlawed

lindsayvallen, 2020-03-12:
Since the season has gone in the crapper, we're looking at adding the sucker chassis despite the official ruling.

lindsayvallen, 2020-03-12:
Merged develop into feature/add_sucker, and did some updating based on new info from JCH (re: one Talon only, ID 5, and don't use sucker when in high gear). Need to test it on robot once hardware is implemented.

WesleyJ-128, 2021-02-19:
Sucker works, but is going to be removed for the 2021 season since it's unnecessary and still illegal.

closed 2021-02-19 by WesleyJ-128