first95/FRC2020 #3: Investigate and fix loop overrun warnings

jwalthour, 2019-10-02:
For some reason, our main loop sometimes takes too long to run.  It's no longer keeping our robot from working, but it could cause us problems in the future.

Possible causes:

- We may have unnecessary CAN interactions.  For example, maybe we are applying PIDF coefficients from the smartdashboard every iteration.
- We should check for any loops doing a lot more things than necessary
- Are we doing any kind of data logging right now?
- Maybe we're sending too many items up to the smartdashboard?

lindsayvallen, 2019-10-30:
Note that there's a comment about this issue in the blog post below where it stresses that this is a warning and not an error, and that it results from a Watchdog that was added in 2019 (hence not seeing it previously). I suspect this mention (especially the bold text) means that a lot of teams may have experienced it this season.

https://wpilib.screenstepslive.com/s/currentCS/m/getting_started/l/1028812-new-for-2019

lindsayvallen, 2020-01-23:
Iceboxing this issue until determine whether it occurs again this year.