first95/FRC2020 #9: Implement PathFinder for 2020

NHDinosaur, 2019-10-02:
JCH has been wanting this for years, so I figured I could do that.

NHDinosaur, 2019-10-25:
There are two code bases out there for FRC pathfinding that seem to be widely used. One which is by Jaci is in working order, but there's another one by team 6135 (Arctos, part of the Ontario district) that doesn't seem to be in complete working order. Therefore, I have decided to use Jaci's pathfinder code and, as of commit 26cc5b9, it has been implemented

lindsayvallen, 2019-10-30:
@NHDinosaur I assume you've seen this documentation, or some better version of it, but I'm adding it here just in case it has some good nugget of info that isn't elsewhere. Also for future reference for other folks.

https://wpilib.screenstepslive.com/s/currentCS/m/84338