first95/FRC2020 #14: Prepare robot breadboard for 2020

lindsayvallen, 2020-01-09:
Take existing breadboard, which is likely not fully functional at the moment, and prepare it for software testing. This should include:

- Swapping the SIM motor(s) for Neo motor(s)
- Swapping the Talon controller(s) for SparkMax controller(s)
- Adding any other components that are missing to get it running

lindsayvallen, 2020-01-09:
This may be something where we want to leverage some non-software folks for assistance.

jwalthour, 2020-01-11:
Additional tasks:
- [ ] Upgrade firmware and image on roboRIO
- [ ] Upgrade firmware on remaining talons, if any are left after upgrading to SparkMaxes
- [ ] Upgrade firmware on SparkMax controllers, if an upgrade is available

lindsayvallen, 2020-01-11:
The bread bot has a robo rio and mini-sims (drive pod) installed, along with a couple of Neos and their SparkMax controllers are in the process of being installed.

zaaj, 2020-01-15:
Used RoboRIO imaging tool to Format Target with FRC_roboRIO_2020_v10.zip
and Update Firmware to check that Firmware version was already 6.0.0 - displayed as 6.0.0f1
Used Phoenix Tuner to update Talon SRX devices from 4.11 to 20.0

Still to do: update Spark MAX controllers

zaaj, 2020-01-15:
Screenshots taken during the update
[image]
[image]
[image]
[image]

zaaj, 2020-01-16:
Followed the firmware update instructions at 
https://www.revrobotics.com/sparkmax-software/#firmware-update-instructions
and updated the firmware on the two Spark MAX's on the Bread Bot
[image]
They were on version 1.4 - newer version includes the USB to CAN bus bridge, so I only needed to plug into the first one.  I set the CAN IDs to 31 and 32, but not sure that will stick after powering down

zaaj, 2020-01-17:
Reviewed CAN connected devices on Bread Bot - checked Cross The Road Electronics website for any firmware updates available for the Pnumatics Control Module and the Power Distribution Panel - they were both on the most recent version available according to Phoenix Tuner

closed 2020-01-18 by jwalthour

zaaj, 2020-01-18:
Posting images to use in Firmware Update markdown here
[image]