first95/FRC2020 #17: General strategy for processing high port vision target

lindsayvallen, 2020-01-10:
Determine a general approach to identify the high port vision target, and where the robot is relative to the target (position and orientation).

lindsayvallen, 2020-02-19:
Useful links for this work

- Properties of OpenCV mats: https://docs.opencv.org/3.4/javadoc/org/opencv/core/Mat.html
- Image processing tutorial that includes useful info on OpenCV moments: https://towardsdatascience.com/computer-vision-for-beginners-part-4-64a8d9856208

Working on this in git branch feature/outer_port_finder.

lindsayvallen, 2020-02-19:
Can use moments to get rotational inertia term John suggested:
I_{1}=\eta _{{20}}+\eta _{{02}}
from https://en.wikipedia.org/wiki/Image_moment

jwalthour, 2020-02-19:
I should probably mention: the OpenCV simple blob detector computes inertial ratio by default.  GRiP doesn't expose it as an option, but you should be able to set it in Java.  Depending how your code is structured it might save you a lot of time.

Here's what setting up a simple blob detector looks like in Python - note the `minInertiaRatio` line.
(from that camera calibration project I mentioned)

````python

    def make_detector(self):
        # Setup SimpleBlobDetector parameters.
        parms = cv2.SimpleBlobDetector_Params()
         
        # Change thresholds
        parms.minThreshold = 0;
        parms.maxThreshold = 128;
         
        # Filter by Area.
        parms.filterByArea = True
        parms.minArea = 5
         
        # Filter by Circularity
        parms.filterByCircularity = True
        parms.minCircularity = 0.25
         
        # Filter by Convexity
        parms.filterByConvexity = False
        parms.minConvexity = 0.9
        parms.maxConvexity = 1
         
        # Filter by Inertia
        parms.filterByInertia = True
        parms.minInertiaRatio = 0.5
        
        # logger.debug("Orig minDistBetweenBlobs: " + str(parms.minDistBetweenBlobs))
        parms.minDistBetweenBlobs = 5
        parms.blobColor = 0
         
        # Create a detector with the parameters
        return cv2.SimpleBlobDetector_create(parms)

````

lindsayvallen, 2020-02-19:
@jwalthour you did mention that Saturday (and possibly in Slack too), hence why I have the mention above about the inertial metric. However, you also mentioned that the blob detector seemed slower than finding lines or contours, so I thought finding the info using something other than the blob detector was worth trying.

jwalthour, 2020-02-19:
great plans!

jwalthour, 2020-03-07:
Today I started a draft version of this that uses the line finder.  I figure it might end up being a useful starting point for later.  I committed it to feature/outer_port_finder, but in separate files, so as to not interfere with the contour-based approach.