first95/FRC2020 #18: General strategy for processing control panel color info

lindsayvallen, 2020-01-10:
Determine a general approach to identify the high port vision target, and where the robot is relative to the target (position and orientation). Suggest using GRiP and example images from FIRST.

jwalthour, 2020-02-15:
Here is my current strategy:

High-level:
- Camera is aimed to look up at the control panel
- Find the region inside the inner ring of the lazy susan
- Find the 8 colored pie slices
- Compute the angle of the color wheel from the relative locations of the pie slices

jwalthour, 2020-02-15:
Example:
[image]

Based on input image:
[image]

jwalthour, 2020-02-15:
Detailed plan:
- Run HSV filter to find colorful parts of the image
- Blur (to fill in dark lines from chromatic abberation)
- Find blobs, with settings so it only finds the ellipse in the center of the lazy susan
- Crop image to that area
- For each color:
    - Run HSV filter to select the appropriate hue
    - Blur
    - Find blobs
- At this point we should have 8 blobs representing all of the pie slices within the center of the lazy susan, two for each color.  However, it's possible there will be something visually in the way (such as glare or a scratch).
- For each pair of colored blobs, compute angle of the line connecting them.  This will give us the azimuth angle of the control panel relative to the robot's heading.  Since we can only get a camera up under the control panel while the robot is relatively square to the front of the control panel, we can treat this as the azimuth angle of the control panel relative to the field.

jwalthour, 2020-02-15:
A note: it appears dramatically faster to find and filter contours than blobs:
[image]