first95/FRC2020 #19: Implement and refine code for high port target

lindsayvallen, 2020-01-10:
After #17 is done, implement the general strategy in Java code and refine it. Implementation will also involve identifying, evaluating, and confirming which hardware (camera and lights) to use. This refinement should include creating a mock high port and taking images of it in controlled ways.

jwalthour, 2020-02-21:
Updates from tonight:
- Students added retroreflective tape to a mock high goal target
- I added stub code in the branch `feature/vision_basics`, which pulls images from the camera within the robot code.  Should be able to drop a processing step into the `VisionProcessing` class, or, replace `VisionProcessing` entirely.

lindsayvallen, 2020-03-12:
Combined this into #17 so closing this one.

closed 2020-03-12 by lindsayvallen