first95/FRC2020 #22: Test updateTo2020 branch and then merge it

lindsayvallen, 2020-01-11:
I created a branch from develop called feature/updateTo2020 that had the Gradle 2020 update done to it (per https://docs.wpilib.org/en/latest/docs/software/wpilib-overview/importing-gradle-project.html) and then necessary changes to get it to build.

- [ ] Update Cecilia's RoboRio and Talons' firmware
- [ ] Push the branch described above to Cecilia and confirm that the main functionality is still working
- [ ] If there are issues relative to how it was at the end of last season, fix them if they're minor OR if they keep the robot from working at all, and make issues for them if they're significant but the robot can at least still move
- [ ] Merge that branch back into develop and notify everyone so they know to update any active feature branches (e.g. pathfinder)

jwalthour, 2020-01-11:
I tried updating the roborio image and firmware on Cecila.  It reported success to the firmware upgrade.  THen i wrote the image.  IT reported "corrupt firmware" but was still able to find the roborio, and reported that the image was the 2020 one.  We'll want to look at this again.

jwalthour, 2020-01-15:
7018b356d78a7e7047bd0338f280b09d4c58a888

closed 2020-01-15 by jwalthour