first95/FRC2020 #25: Manage Grasshoppers' Slack storage

lindsayvallen, 2020-01-18:
Our Slack is running out of space (see image below). I'm guessing this is largely because of the many videos we've posted recently. I suggest we come up with an alternative method for storing/sharing large files, e.g. keep in a Google account and link to Slack, or create a Dropbox account and share through that. Get coach buy-in on whatever do, and then transfer files.

[image]

jwalthour, 2020-01-18:
@jch13

jwalthour, 2020-01-29:
I haven't found any easy way to sort files by size or anything like that.  But we can pull up a list of files here and start deleting, if we ever need to:

https://app.slack.com/client/T74UGMJGG/C75KVKUUU/files

lindsayvallen, 2020-03-12:
Looks like we deleted a bunch of files that reduced the overall storage use from ~5GB (limit) to ~3GB (current) based on this: https://ghoppers.slack.com/admin/stats Hence, I'm going to say this issue is closed, and we can re-open it later, if needed.

closed 2020-03-12 by lindsayvallen