first95/FRC2020 #27: Add Git commit or hash to what's printed when robot is run

lindsayvallen, 2020-01-18:
When we run the robot, it would be great to have the Git info (e.g. Git status) printed to the driver station or Smart Dashboard so we know what's on the robot when we run it, not just by re-pushing code to it.

lindsayvallen, 2020-01-18:
@jwalthour suggested that we have something that gets 'git status', saves it to a file, and pushes that to the robot during the code build process.

zaaj, 2020-01-22:
First find, haven't dug into it yet, but looks like it will facilitate what we want:
Git-Version Gradle Plugin
https://github.com/palantir/gradle-git-version

zaaj, 2020-01-23:
Challenge I was trying to figure out is how to get a git version, which is a hash of the source code, into the source code, since altering the code after calculating the hash invalidates the hash.  The only thought I came up with on my own was to get Gradle to create a small java file that would be gitignore'd, yet compiled in, which had the sole purpose of providing a public final static string of the version to the project.

However, there appears to be cleaner method.  Java can reference attributes stored in the manifest portion of the Jar file, and Gradle can add attributes to the manifest. 

A stack overflow answer lays it out nicely:  [Gradle: Make build version available to Java](https://stackoverflow.com/questions/37642837/gradle-make-build-version-available-to-java)
 
So we would add the Git-Version Gradle Plugin, then use it to define the Implementation-Version attribute to contain the git version information.  That attribute can be read within java code with
`BuildVersion.class.getPackage().getImplementationVersion();`

Full git status info is an interesting idea I'll have to keep digging for.

riffing on the Manifest attribute theme, I searched for 
[How do I add a Build-Host attribute to a JAR manifest using Gradle?](https://stackoverflow.com/questions/8930002/how-do-i-add-a-build-host-attribute-to-a-jar-manifest-using-gradle)
to find that Gradle doesn't have a public API for the build host, but that it's available in Java using
`InetAddress.localHost.hostName`

Haven't figured out if Gradle can call java, or if gradlew.bat could be used to pass in the output of the hostname command.

zaaj, 2020-01-25:
Branch origin/feature/add-git-info-to-JarManifest now has an updated build.gradle which adds the plug-in line:

zaaj, 2020-01-30:
Added the two attributes from the Jar Manifest to the SmartDashboard
[image]
(you can even see which commit ID it was done in ;)

closed 2020-01-30 by zaaj