first95/FRC2020 #28: Confirm current understanding of robot systems and plan for executing

lindsayvallen, 2020-01-22:
I talked with JCH and wrote [this Wiki page](https://github.com/first95/FRC2020/wiki/Robot-System-Description) based on that conversation. We need to confirm this understanding is correct, and then plan how to execute the associated software including relative priorities (e.g. which systems first) and what's needed for the different systems (e.g. list of commands).

lindsayvallen, 2020-01-22:
@jch13 @jwalthour @NHDinosaur @zaaj @WesleyJ-128 please take a look at the Wiki page I linked in the issue description above. Does that match your understanding? Is there anything missing? Is there anything high-level that we need to clarify? Feel free to make/save comments directly in the Wiki page.

jwalthour, 2020-01-23:
Looks good to me - matches my current understanding

jwalthour, 2020-03-04:
Done

closed 2020-03-04 by jwalthour