first95/FRC2020 #39: Get sensors 1-3 working from code

lindsayvallen, 2020-02-14:
Sensor 0 (at singulator) shows up as toggling in Smart Dashboard when the sensor is covered/uncovered. However, none of the other sensors (1-3) show any change in SmartDashboard even when the lights on the physical sensors change. @zaaj said sensor 0 is a different physical model than the others, so maybe they need some tweaks to how they're used.

zaaj, 2020-02-14:
James set up a multimeter, we confirmed sensor 0 uses the white cable, and the different-model sensors 1 - 3 use black.  Team is working on re-wiring 1 - 3.

zaaj, 2020-02-20:
Ended up that a Voltage Divider was needed for sensors 1 - 3.  Once those were soldered in, sensors started working in code.  Sensor 0 provided reversed logic, and only needed a current limiting resistor in-line.  getSingulatorSensor updated to reverse the logic.

closed 2020-02-20 by zaaj