first95/FRC2020 #42: Decide how shooter should interact with PowerCellMover

lindsayvallen, 2020-02-14:
Right now shooter is a sub-system of its own. Need to decide if it should be integrated with PowerCellMover, in part because need to run indexer while shooting.

lindsayvallen, 2020-02-15:
Should integrate shooter with rest of power cell handler.

jwalthour, 2020-02-20:
Inputs:
- 2 user controls
    - Fire button
    - Intake button
- 4 sensors:
    - Singulator occupied (by 1 or more balls)
    - Indexer entrance occupied
    - Indexer position 1 occupied
    - Shooter loaded

Outputs:
- Speed of ground intake rollers
- Speed of singulator intake rollers
- Speed of singulator 
- Speed/position of indexer
- Speed of shooter flywheel

jwalthour, 2020-02-21:
Whoops, I think I misread the scope of the issue - the above comment might not be quite right

lindsayvallen, 2020-03-12:
Shooter has been integrated into power cell handler so closing this issue.

closed 2020-03-12 by lindsayvallen