first95/FRC2020 #46: Get more computers working and sequester ones that don't

lindsayvallen, 2020-03-12:
We have plenty of computers in various states of maintenance, repair, or lack thereof, and little clear indication of which is which. For each computer, we should identify which of the following categories it falls into:

- Software development: good quality laptop, has at least most of the SW dev tools on it that are relatively up-to-date
- General use: reasonable quality laptop that can be powered on, has a charger that works, can be logged into, and reliably connect to the internet for OnShape, Slack, part ordering, etc.
- Maybe salvageable: modern-ish looking laptop that can be powered on and has a charger that works, but may not have a log-in we know of or may have internet connection issues
- Should get rid of: despite reasonable efforts, can't shift it into the "Maybe salvageable" category

We should document the status of each computer (in this issue? Maybe also with color-coded tags on the machines?), and try to bump them up into a higher category. For those that can't get moved out of "Should get rid of", we should get rid of them.

lindsayvallen, 2020-03-12:
@jwalthour @zaaj if either of you has some time and are looking for something to do, this would be super helpful so that we winnow down our computers to ones that work and can determine if we need any more, and if so, of what caliber.

zaaj, 2020-03-21:
Well, I took home one of the donateed (non hactc domain member) Lenovo laptops to try updating it to Windows 10.  The Lenovo updater popped up, and offered a high-priority BIOS update, among other.  I ran the BIOS update in case the updater wouldn't be available for Windows 10 (sometimes the case with older hardware) and the updater blue-screen-crashed while updating the BIOS, and now it won't even get to a BIOS setup screen.  I guess Winnow is one word for that process :-(